QuickCorp/pyGT511C3 #4: Raspberry Pi Issues

louishudd, 2015-01-06:
Hi am wondering if you can help me.

I have been trying to get this to work on my Raspberry Pi. The scanner is plugged into the uart pins.

Enroll code:
```
import FPS, sys, time

if __name__ == '__main__':
	fps = FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=9600,timeout=2,is_com=False)
	fps.SetLED(True)
	fps.UseSerialDebug = False
	if (fps.IsPressFinger() == True):
		print 'finger is pressed'
		start = fps.EnrollStart(1)
		print start
		time.sleep(1)
		fps.CaptureFinger(True)
		e1 = fps.Enroll1()
		print e1
		time.sleep(1)
		fps.CaptureFinger(True)
		e2 = fps.Enroll2()
		print e2
		time.sleep(1)
		fps.CaptureFinger(True)
		e3 = fps.Enroll3()
		print e3
	fps.Close()

```
This all returns 0

Check number enrolled code:
```
import FPS, sys, time

if __name__ == '__main__':
	fps =  FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=9600,timeout=2,is_com=False)
	print fps.GetEnrollCount()
```
This also returns 0

jeanmachuca, 2015-01-07:
You must to wait until isPressFinger ==True with delay 1 second:
           while not fps.IsPressFinger():
                FPS.delay(1)

Check the sample:

https://github.com/QuickGroup/pyGT511C3/blob/master/LegacyEnroll.py

louishudd, 2015-01-07:
When I run LegacyEnroll.py I get:
```

Command: CheckEnrolled
55 aa 01 00 50 00 00 00 21 00 71 01
U▒P!q
U\xaa\x01\x00P\x00\x00\x00!\x00q\x01
readed: 55 aa 01 00 04 10 00 00 31 00 45 01
Command: CheckEnrolled
55 aa 01 00 51 00 00 00 21 00 72 01
U▒Q!r
U\xaa\x01\x00Q\x00\x00\x00!\x00r\x01
readed: 55 aa 01 00 04 10 00 00 31 00 45 01
Command: CheckEnrolled
55 aa 01 00 52 00 00 00 21 00 73 01
U▒R!s
U\xaa\x01\x00R\x00\x00\x00!\x00s\x01
readed: 55 aa 01 00 04 10 00 00 31 00 45 01
Command: CheckEnrolled
55 aa 01 00 53 00 00 00 21 00 74 01
U▒S!t
U\xaa\x01\x00S\x00\x00\x00!\x00t\x01
readed: 55 aa 01 00 04 10 00 00 31 00 45 01
Command: CheckEnrolled
55 aa 01 00 54 00 00 00 21 00 75 01
U▒T!u
U\xaa\x01\x00T\x00\x00\x00!\x00u\x01
readed: 55 aa 01 00 04 10 00 00 31 00 45 01
Command: CheckEnrolled
55 aa 01 00 55 00 00 00 21 00 76 01
U▒U!v
U\xaa\x01\x00U\x00\x00\x00!\x00v\x01
readed: 55 aa 01 00 04 10 00 00 31 00 45 01
```
And nothing happens when I put my finger on the reader

jeanmachuca, 2015-01-07:
Check the baud rate of your device. Probably is 115200 and not 9600. Test a major delay than 1
 If nothing happens yet, you must review with another device

louishudd, 2015-01-08:
Running: 
```
print fps.ChangeBaudRate(9600)
```
Returns:
```

Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
U▒
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
readed: 55 aa 01 00 00 00 00 00 30 00 30 01 5a a5 01 00 10 04 14 20 00 00 00 00 ef 13 ef 40 14 c8 63 34 12 97 81 1f 27 00 00 00 5c 06
readed:
False
```

I have also tried setting the delay to 5 seconds

jeanmachuca, 2015-01-09:
Please compare fps._lastResponse.Error value with the error table un the datasheet

also, in fps._lastResponse.errors are a Key value definition for common error supported

Please tell me your result!

louishudd, 2015-01-12:
Sorry but I don't understand. Where do I find this fps._lastResponse.Error value?

jeanmachuca, 2015-01-12:
after your command, example:

print fps.ChangeBaudRate(9600)
print fps._lastResponse.Error

and then compare

louishudd, 2015-01-12:
It just returns 
```
INVALID
```

jeanmachuca, 2015-01-13:
So there is a major trouble, 

Are you sure you are using GT511C3 device and not GT511C1 or another variant? 
You are using the device connected directly to the GPIO? or there are another device connected in the middle? For example, with an FTDI UART Card connected to USB, this device is not compatible with 9600 baudrate, and you need to change the baud to 115200

I think the trouble is in the response of your device, or in your device, is a hardware problem

I don't see how I can help you to much...

Please tell me if you get more results in your test, thanks

louishudd, 2015-01-20:
Thanks I will let you know how it goes. Have a coffee on me @jeanmachuca @ChangeTip !

changetip, 2015-01-20:
Hi @jeanmachuca, @louishudd sent you a Bitcoin tip worth a coffee (7,082 bits/$1.50), and I'm here to deliver it ➔ **[collect your tip](https://www.changetip.com/collect/367110)**.

**[Learn more about ChangeTip](https://www.changetip.com/tip-online/github)**

mikemanosky, 2015-02-19:
Hi there, has this issue been fixed as I'm getting ths problem. Please could Jean test this code himself with his scanner to check it still works with the recent modifications on the COM ports please? Thanks. Once the baud rate is changed the scanner refuses to respond to packets.

louishudd, 2015-02-23:
@mikemanosky It has not yet been fixed. I don't have the time to try anything else at the moment, let me know if you find anything.

naingyeaung, 2015-05-05:
I am using a GT511C1 and I understand this is written for GT511C3. I am wondering if running "raspberry_pi_GPIO_test.py" should turn the Control CMOS LED on regardless of the variant. Any advice? Thanks!

massivedynamics, 2015-06-28:
I worked with C1. It works. For enroll proccess it's better you use "while" instead of "if" with finger pressed function. Maybe you want to look results of enroll process to get out of while for stable enrollment.

ribs85, 2015-07-11:
I'm having the same issues with my raspberry pi: The library claims my finger is on the sensor nearly all the time, and I haven't been able to enroll a single fingerprint yet.

The blue LED comes on, so I know there is communication.

Laughably, the library claims my finger is on the scanner even when I don't have it plugged in. The error message ranges from "None" to "INVALID".

I'm guessing something fairly deep (and probably simple) is wrong here, but I have no idea where to start fixing it.

jpatryk, 2015-11-26:
Hi,
what i can grab the finger and save into external sql database?

sankarmanoj, 2016-02-15:
If you are having trouble with a Raspberry Pi, and you are using the GPIO Serial Pins on the Raspberry Pi, you need to disable the Serial Terminal. 

Open your terminal and type `sudo rapsi-config`. Go to Advanced Settings, where you should see something regarding Serial. Click on it and disable Serial. Reboot your Raspberry Pi. This might fix some issues.

himanshu-hearthacker, 2016-07-10:
@louishudd  Did u solved that problem of INVALID, i do not get any response from fingerprint scanner,
but it worked fine with windows software. I am confused is there any problem in hardware or software.

mikemanosky, 2016-07-10:
@himanshu-hearthacker I'm having exactly the same problem. Windows software worked fine but as soon as I try this library nothing works and haven't been able to store one fingerprint or recognize one through it. Frustrating.

sankarmanoj, 2016-07-10:
If you are using the RPi's serial pins, make sure that you have disabled
the console over Serial. If it still does not work, try out the Node Js
library for the same. I have had success using the Nodejs library

mikemanosky, 2016-07-10:
Im trying to use a FTDI device to link into the usb of my PC, the light turns on to scan a fingerprint and shows a connection but it refuses to do anything else after that

massivedynamics, 2016-07-10:
Most of communication error is about timing. Add print to send and get part to track hex response.

himanshu-hearthacker, 2016-07-10:
my program got struck at CHECKEnrolled() function, it returns False every time, and i print fps._lastResponse.Error statement but it always return INVALID, my serial are disable as well. I change my Baud rate and delay many times . If i removed CHECKErolled() function,  Sometime it recognise my finger for first time but not more than that.

himanshu-hearthacker, 2016-07-20:
hi,
 Use FPS2.py file in your code from [https://github.com/Jucker/SIPIntercom-with-FPS]. that guy uses same code with some modifications. it worked fine for 2 enrolment then returned with NACK_TURN_ERR.
If you resolved with it. please help me out.
Use baud rate 115200

himanshu-hearthacker, 2016-07-21:
get it done ...some times creates problem but it is working now