QuickCorp/pyGT511C3 #7: Connection to Fingerprint Scanner from FTDI Device

mikemanosky, 2015-02-19:
Hi there,
Sorry to be a pest but it seems like the variable DEVICE_NAME is not found through the class (FPS_GT511C3) because it is not set unless you pass it in as COM3. Could you check for me that this code still works for a COM3 device please? It doesn't seem to change to a 115200 Baudrate for me.

Thanks Michael

jeanmachuca, 2015-02-19:
Hi Michael,

DEVICE_NAME variable is COM3 for nt (Windows) systems,  and /dev/cu.usbserial-A601EQ14 for other (linux, mac, etc with usb serial cable)

For to change the baud rate to 115200 use the sample code in:

https://github.com/QuickGroup/pyGT511C3/blob/master/baud_to_115200.py

Yo can get a connected device list with the devices() method in FPS, just type in shell:
> import FPS
> print FPS.devices()

In your code, for to connect another custom device, set DEVICE_NAME adding the following two line code first:

import FPS
FPS.DEVICE_NAME = 'CustomDEVICE' # your custom device in devices() list

good luck

mikemanosky, 2015-02-23:
Thank You
However I still cant get the code to work with a Baudrate of 115200. The device refuses to respond after running the baud_to_115200.py file. Is this an error in the code? 
Thanks Michael

jeanmachuca, 2015-05-17:
Just try connecting FTDI to another usb port, I have been testes this device in both cases and no troubles for me