QuickCorp/pyGT511C3 #8: Works only with 9600 baudrate

Lucamenghini, 2015-03-09:
Hi I've bought a GT511C3 and a GT511C1R.
I've tried to connect them to my Raspberry PI B+.
With your code and with my code (in C# with mono) the only working baudrate is 9600.
My goal is to obtain a scanned fingerprint or a Template and transfer it to the raspberry.
No enrolling is needed.
With serial in 9600 baudrate I can set command OPEN and turn on the led and recognize when  a finger is on the sensor perfctly. But the command CaptureFinger doesn't work (it give no response).
I wrote on the arduino forums some arduino user resolve it using 115200 baudrate.
But with my Raspberry it doesn't work. I set baudrate to 115200 and open + turn on led don't work.

So I tried the command GetRawImage (after a verified IsPressFinger) with 9600 baudrate but i't doesn't work. Sometimes I retrieved a lot of bit for a few seconds (Hurra! it's my fingerprint!) but it stopped immediately (at 9600 the raw image it must be retrieved in no less time than 30seconds).

No problem! I tested the GetRawImage at 115200... but it is impossible to Open and to turn on the led...
Nothing....
The Raspberry appears to be limited to 9600.
Why Arduino users use the GT511c3 al 115200 with no problem?
Have you tested it?

Bye 

Luca Menghini

mikemanosky, 2015-03-17:
Hi there, I have experienced the same issue. It just doesn't work with the 115200 baudrate. I have tried changing the code to work but I don't know enough about it sadly. Please let me know if you manage to get it working with the 115200 baud rate. Thanks :)

Lucamenghini, 2015-03-22:
Hi Michael. Somewhere I wrote that it is necessary to start it at 9600 and command a change baud rate to 115200 after. But it's not clear how can i manage a parallel variation of speed with the serial.
Is it possible to start the serial at 9600 and to start the fps at 9600 and then command a change at the fps at 115200 and then change the serial to 115200?
Or start the serial at 115200 and the fps at 9600 and command a change to 115200.
I think in both the situation when the baud are different between serial and fps it not possible to send packet to the fps.
Or not?

Inviato da iPhone5

Lucamenghini, 2015-04-30:
Solved!
Declare 2 different serials. First at 9600 and second at 115200.
Open the first, set command ti open FPS, set command ti change baudrate of FPS.
The n open the second serial and close the first.
Clear the serial channel now.
Verify when pressfinger return true, command a capture and then a getimage.
That's all
It work!!!

jeanmachuca, 2015-04-30:
All right!! Congrats!

closed 2015-04-30 by jeanmachuca