QuickCorp/pyGT511C3 #10: no response packet

naingyeaung, 2015-05-15:
I am trying to run the code on raspberry pi. When I ran "raspberry_pi_GPIO_test.py". The response packet is empty. The details of the result is added below:

pi@raspberrypi ~/pyGT511C3-master $ sudo python raspberry_pi_GPIO_test.py
Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
Uª
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
readed:
Command: CmosLed
55 aa 01 00 01 00 00 00 12 00 13 01
Uª
U\xaa\x01\x00\x01\x00\x00\x00\x12\x00\x13\x01
readed:
Put your finger in the scan
Command: IsPressFinger
55 aa 01 00 01 00 00 00 26 00 27 01
Uª&'
U\xaa\x01\x00\x01\x00\x00\x00&\x00'\x01
readed:
Your finger is in the scan
Command: CmosLed
55 aa 01 00 00 00 00 00 12 00 12 01
Uª
U\xaa\x01\x00\x00\x00\x00\x00\x12\x00\x12\x01
readed:
Command: Close
55 aa 01 00 00 00 00 00 02 00 02 01
Uª
U\xaa\x01\x00\x00\x00\x00\x00\x02\x00\x02\x01
readed:

Would you mind pointing out the possible issues? Thanks!

jeanmachuca, 2015-05-17:
Hi,

How do you have connected the device, through an FTDI card or GPIO directly?

naingyeaung, 2015-05-17:
Hi,

        I connected through GPIO.

massivedynamics, 2015-06-28:
Response wait is too low. I add extra delay for it to get response.