QuickCorp/pyGT511C3 #11: isPressFinger() returns always True

m-hertig, 2015-05-25:
Hi, thank you for this great library. I could manage to sometimes download the raw Image from the scanner using the functions you provided in test_raw.py. However, it only works once in many times and isPressFinger() returns always true. What am I doing wrong? Here an examples from commands I typed into the python shell:

    >>> fps =  FPS.FPS_GT511C3(device_name='/dev/cu.usbserial-A403CT1J',baud=115200,timeout=2,is_com=False)
    Command: Open
    55 aa 01 00 01 00 00 00 01 00 02 01
    U?
    U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
    readed: 
    >>> 
    >>> fps.SetLED(True)
    Command: CmosLed
    55 aa 01 00 01 00 00 00 12 00 13 01
    U?
    U\xaa\x01\x00\x01\x00\x00\x00\x12\x00\x13\x01
    readed: 
    False
    >>> fps.IsPressFinger()
    Command: IsPressFinger
    55 aa 01 00 01 00 00 00 26 00 27 01
    U?&'
    U\xaa\x01\x00\x01\x00\x00\x00&\x00'\x01
    readed: 
    True
    >>> fps.IsPressFinger()
    Command: IsPressFinger
    55 aa 01 00 01 00 00 00 26 00 27 01
    U?&'
    U\xaa\x01\x00\x01\x00\x00\x00&\x00'\x01
    readed: 
    True

jeanmachuca, 2015-05-25:
Hi, you are welcome to use my code to your own purposes!
It appears there are nothing wrong in your test code...

You had tested with another timeout setting calling FPS.FPS_GT511C3(...)?

If no, set another greater or lower value for this and test again 

If yes and no result... Yo must probably review your PIN connections or your USB connection as well

In my case, most troubles connecting this device through an FTDI card was because the USB serial cable. I changed this and then all works fine... 

Tell me if my answer is useful for you and your results!

Regards,

m-hertig, 2015-05-27:
Thank you for you reply! I got rid of the USB FTDI and tested it on the Raspberry Pi pins  (ttyAMA0). The problems are similar; It is possible to connect it with 9600 baud and to switch the led on and off, but as soon as I want to switch to 115200 baud, it won't work anymore. I typed every command separately in the python shell, so there is a quite big delay between the commands. Do you see what I am doing wrong? Also I'd like to know; is it correct that the baud rate must be changed not only once, but once per boot or every time that you plug in the sensor?

    >>> fps =  FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=9600,timeout=5,is_com=False)
    Command: Open
    55 aa 01 00 01 00 00 00 01 00 02 01
    U?
    U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
    readed: 
    >>> fps.SetLED(True)
    Command: CmosLed
    55 aa 01 00 01 00 00 00 12 00 13 01
    U?
    U\xaa\x01\x00\x01\x00\x00\x00\x12\x00\x13\x01
    readed: 
    False
    >>> fps.SetLED(False)
    Command: CmosLed
    55 aa 01 00 00 00 00 00 12 00 12 01
    U?
    U\xaa\x01\x00\x00\x00\x00\x00\x12\x00\x12\x01
    readed: 
    False
    >>> fps.ChangeBaudRate(115200)
    Command: ChangeBaudrate
    55 aa 01 00 00 c2 01 00 04 00 c7 01
    U???
    U\xaa\x01\x00\x00\xc2\x01\x00\x04\x00\xc7\x01
    readed: 
    False
    >>> fps.Close()
    Command: Close
    55 aa 01 00 00 00 00 00 02 00 02 01
    U?
    U\xaa\x01\x00\x00\x00\x00\x00\x02\x00\x02\x01
    readed: 
    False
    >>> fps =  FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=115200,timeout=5,is_com=False)
    Command: ChangeBaudrate
    55 aa 01 00 80 25 00 00 04 00 a9 01
    U??%?
    U\xaa\x01\x00\x80%\x00\x00\x04\x00\xa9\x01
    readed: 
    Command: Open
    55 aa 01 00 01 00 00 00 01 00 02 01
    U?
    U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
    readed: 
    >>> fps.SetLED(True)
    Command: CmosLed
    55 aa 01 00 01 00 00 00 12 00 13 01
    U?
    U\xaa\x01\x00\x01\x00\x00\x00\x12\x00\x13\x01
    readed: 
    False

sankarmanoj, 2016-02-15:
If you are having trouble with a Raspberry Pi, and you are using the GPIO Serial Pins on the Raspberry Pi, you need to disable the Serial Terminal. 

Open your terminal and type `sudo rapsi-config`. Go to Advanced Settings, where you should see something regarding Serial. Click on it and disable Serial. Reboot your Raspberry Pi. This might fix some issues.

ribs85, 2016-03-03:
So what it's worth, turning off that serial setting (which I think was already off for me) didn't make any difference.

ramyeid, 2016-04-07:
hello did someone fix this?
I have the same problem my connections are correct and i tried with different timeout values.