QuickCorp/pyGT511C3 #12: Issue with Raspberry Pi

nislinux, 2015-09-11:
Dear Sir,
Im using **GT-511C3** with **Raspberry Pi 2**.
FS is connected in **GPIO**s of Pi.
I made all the connections properly.

Step 1
---------

When I tried to run : **sudo python raspberry_pi_GPIO_test.py**
Im getting the following messages -

*Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
Traceback (most recent call last):
  File "raspberry_pi_GPIO_test.py", line 32, in <module>
    fps =  FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=9600,timeout=2,is_com=False)
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 319, in __init__
    self.Open()
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 334, in Open
    rp = self.GetResponse()
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 775, in GetResponse
    r = bytearray(self._serial.read(self._serial.inWaiting()))
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 449, in inWaiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
IOError: [Errno 5] Input/output error*

-------------------------------------------------------------------------
Step 2
-----------------------------------------------------------------------------
2 .Then I tried  **sudo python baud_to_115200.py**
result was -
Traceback (most recent call last):
  File "baud_to_115200.py", line 16, in <module>
    fps = FPS.FPS_GT511C3()
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 316, in __init__
    self._serial = connect(device_name,baud,timeout,is_com=is_com)
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 277, in connect
    device_name = DEVICE_NAME
UnboundLocalError: local variable 'DEVICE_NAME' referenced before assignment

------------------------------------------------------------------------------------
Step 3
---------------------------------------------------------------------------------
 **sudo python LegacyEnroll.py**
result was -

Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
Traceback (most recent call last):
  File "LegacyEnroll.py", line 91, in <module>
    fps = FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=9600,timeout=2,is_com=False) #settings for raspberry pi GPIO
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 320, in __init__
    self.Open()
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 335, in Open
    rp = self.GetResponse()
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 776, in GetResponse
    r = bytearray(self._serial.read(self._serial.inWaiting()))
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 449, in inWaiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
IOError: [Errno 5] Input/output error

---------------------------------------------------------------------------
I came to know the problem may be with the Baud rate.
So I **changed the baud** rate in the **LegacyEnroll.py** to 115200
**fps = FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=115200,timeout=2,is_com=False)**

[... 181 lines cut ...]
  File "/home/pi/Project/pyGT511C3-master/FPS.py", line 277, in connect
    device_name = DEVICE_NAME
UnboundLocalError: local variable 'DEVICE_NAME' referenced before assignment

-----------------------------------------------------------------------------------------
Step 6
------------------------------------------------------
As a try, I made a **small change** in **enrolar.py**
*#fps = FPS.FPS_GT511C3()
    fps = FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=115200,timeout=2,is_com=False)*

and run again, the result was - 
Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
readed: 
Traceback (most recent call last):
  File "enrolar.py", line 13, in <module>
    Enroll(fps,sys.argv[1])
IndexError: list index out of range

--------------------------------------------
What may be the issue ?
Please help.

DimaSemchuk, 2016-01-03:
I have kind the same issue with slightly different errors

raufbulbul, 2016-12-17:
Same problem for me. Is there anybody for answer ???

raufbulbul, 2016-12-22:
???