QuickCorp/pyGT511C3 #13: Identify1_N returns 0

cheatcodex, 2015-09-25:
Hi Jean,
I've been working with this device for a couple of weeks now. Can't figure out the last and final step, the identification part. 
I noticed there's a test_raw.py, but I think the default function comes with the device, should do the identification job right? 
I'm calling the function of Identify1_N(). I started with serial timeout=1.0 and the actual identification takes time. I added a 2.5 seconds of delay, right before the GetResponse() inside of the identify1_N() function. The responses are either "55 aa 01 00 00 00 00 00 30 00 30 01" or 31 (not acknowledged). In the 30 ACK event, the ID appears to be 0 when I use any finger on the scanner. 
Would you have any advice? 
Thanks a lot!

cheatcodex, 2015-09-25:
Also in the Legacyenroll.py file, do you think that is a bug with the okid? before it was pointing at 0 the whole time. The checkenrolled returns true if the space is occupied. I think you have it the other way around.

ribs85, 2015-09-27:
How are you connecting? Raspberry pi?

It seems to me that the library simply doesn't work for pi users. I have to write for loops to read the fingerprint as the first one or two attempts *always* fail. Even after getting the 3 enrol commands through, and it apparently working, I cannot ever get it to verify. I always get 0 or 200 as the ID. I managed to get it to 'verify' my chin once, after I apparently enrolled successfully on one of my fingers. I once even managed to get it to verify as zero when the GT511C3 wasn't even plugged in.

I'm thinking I might try connecting it to a Windows PC and using the 'official' SDK, limited as it is. I think there is either a problem with this library, or a problem with how the raspberry pi communicates with the device. Sadly, this is somewhat beyond my skills to debug :(

cheatcodex, 2015-09-28:
Yes. I'm connecting to a Pi. make sure you have GT511C3 since I can only speak for what I've been using.
Always check your pin connections: 
GT-511C3-----------Pi B
1(TX)-----------------10 (RX)
2(RX)-----------------8   (TX)
3(GND)---------------4   (GND)
4(3.3V)----------------1  (3V3)
[image]

I've changed the okid part of the Legacyenroll.py
The problem I had was the code always tries to enroll in id0. I had the problem with identification. I get ACK and ID 0. Then I enrolled another two fingers, and I realized that I still get ACK and ID=0. That's what got me to start look at the Legacyenroll.py I think the okid remains at 0 with the if statement. I changed that, so Legacyenroll.py works for me after another few other fixes. Legacyenroll works but I just found it hard to use so I added a few delays and stuff. 
One thing I didn't have time to figure out was the ChangeBaudRate(self) function. It never worked for me, so I just insert the baud rate I want to use in the open connection function. Changing from 9600 to 115200 in program never works for me. I didn't have time to look into it yet. 

Again, my main problem is that the Identify1_N() always returns ACK with parameter=0 or NACK with random parameter. I noticed the test_raw uses numpy and other libraries, i'm just trying to stay within the chip's functionalities. 

'''
Created on 08/04/2014

@author: jeanmachuca

SAMPLE CODE:

This script is a test for device connected to GPIO port in raspberry pi

For test purpose:

Step 1:
Connect the TX pin of the fingerprint GT511C3 to RX in the GPIO

Step 2:
Connect the RX pin of the fingerprint GT511C3 to TX in the GPIO

Step 3:
Connect the VCC pin of the fingerprint GTC511C3 to VCC 3,3 in GPIO

Step 4:
Connect the Ground pin of fingerprint GT511C3 to ground pin in GPIO

This may be works fine, if don't, try to change the fingerprint baud rate with baud_to_115200.py sample code

This script Enrolls your finger in the device internal database
you have 200 ids avalilables for enroll
Each time you executes this enroll script, enrollid is autoincrement for a free number

'''
import FPS, sys

def LegacyEnroll(fps):
    '''
    Enroll test
    '''
    enrollid=0
    taken_id=True
    #search for a free enrollid, you have max 200
    while taken_id and (enrollid < 200):
        fps.UseSerialDebug=False
        taken_id = fps.CheckEnrolled(enrollid)
        FPS.delay(2.5)
        if taken_id:
            print "id %s is taken /n", enrollid
            enrollid+=1

    if enrollid <200:
        #press finger to Enroll enrollid
        print '@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@Press finger to Enroll %s' % str(enrollid)
        fps.EnrollStart(enrollid)
        while not fps.IsPressFinger():
            FPS.delay(1)
        iret = 0
        if fps.CaptureFinger(True):
            FPS.delay(0.5)
            iret=fps.Enroll1()
            if iret>0:
                print"!!!!!!!!!!!!Error Code: %s",iret
            FPS.delay(2)
            #remove finger
            print 'remove finger'
            FPS.delay(2)
            while fps.IsPressFinger():
                FPS.delay(1)
                print 'Remove Finter'
            while not fps.IsPressFinger():
                FPS.delay(1)
                print 'press same finger again'
            if fps.CaptureFinger(True):
                FPS.delay(0.5)
                iret=fps.Enroll2()
                if iret>0:
                    print "!!!!!!!!!!!!! Error code in round 2: %s",iret
                FPS.delay(2)
                #remove finger
                print 'remove finger'
                FPS.delay(2)
                while not fps.IsPressFinger():
                    FPS.delay(1)
                    #Press same finger again
                    print 'press same finger yet again'
                while not fps.IsPressFinger():
                    FPS.delay(1)
                if fps.CaptureFinger(True):
                    FPS.delay(1)
                    iret = fps.Enroll3()
                    #remove finger
                    print "remove finger"
                    if iret == 0:
                        print 'Enrolling Successfull'
                    else:
                        print 'Enrolling Failed with error code: %s' % str(iret)
                else:
                    print 'Failed to capture third finger'
            else:
                print 'Failed to capture second finger'
        else:
            print 'Failed to capture first finger'
    else:
        print 'Failed: enroll storage is full'

if __name__ == '__main__':
    #changed baud rate to 115200. 09032015
    fps = FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=115200,timeout=2.0,is_com=False) #settings for raspberry pi GPIO
#    fps.Open()
    fps.UseSerialDebug=False
    FPS.delay(2.5)
    if fps.SetLED(True):
        LegacyEnroll(fps)

cheatcodex, 2015-09-28:
'''
Created on 08/04/2014

@author: jeanmachuca

SAMPLE CODE:

This script is a test for device connected to GPIO port in raspberry pi

For test purpose:

Step 1:
Connect the TX pin of the fingerprint GT511C3 to RX in the GPIO

Step 2:
Connect the RX pin of the fingerprint GT511C3 to TX in the GPIO

Step 3:
Connect the VCC pin of the fingerprint GTC511C3 to VCC 3,3 in GPIO

Step 4:
Connect the Ground pin of fingerprint GT511C3 to ground pin in GPIO

This may be works fine, if don't, try to change the fingerprint baud rate with baud_to_115200.py sample code

This script Enrolls your finger in the device internal database
you have 200 ids avalilables for enroll
Each time you executes this enroll script, enrollid is autoincrement for a free number

'''
import FPS, sys

def LegacyEnroll(fps):
    '''
    Enroll test
    '''
    enrollid=0
    taken_id=True
    #search for a free enrollid, you have max 200
    while taken_id and (enrollid < 200):
        fps.UseSerialDebug=False
        taken_id = fps.CheckEnrolled(enrollid)
        FPS.delay(2.5)
        if taken_id:
            print "id %s is taken /n", enrollid
            enrollid+=1

    if enrollid <200:
        #press finger to Enroll enrollid
        print '@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@Press finger to Enroll %s' % str(enrollid)
        fps.EnrollStart(enrollid)
        while not fps.IsPressFinger():
            FPS.delay(1)
        iret = 0
        if fps.CaptureFinger(True):
            FPS.delay(0.5)
            iret=fps.Enroll1()
            if iret>0:
                print"!!!!!!!!!!!!Error Code: %s",iret
            FPS.delay(2)
            #remove finger
            print 'remove finger'
            FPS.delay(2)
            while fps.IsPressFinger():
                FPS.delay(1)
                print 'Remove Finter'
            while not fps.IsPressFinger():
                FPS.delay(1)
                print 'press same finger again'
            if fps.CaptureFinger(True):
                FPS.delay(0.5)
                iret=fps.Enroll2()
                if iret>0:
                    print "!!!!!!!!!!!!! Error code in round 2: %s",iret
                FPS.delay(2)
                #remove finger
                print 'remove finger'
                FPS.delay(2)
                while not fps.IsPressFinger():
                    FPS.delay(1)
                    #Press same finger again
                    print 'press same finger yet again'
                while not fps.IsPressFinger():
                    FPS.delay(1)
                if fps.CaptureFinger(True):
                    FPS.delay(1)
                    iret = fps.Enroll3()
                    #remove finger
                    print "remove finger"
                    if iret == 0:
                        print 'Enrolling Successfull'
                    else:
                        print 'Enrolling Failed with error code: %s' % str(iret)
                else:
                    print 'Failed to capture third finger'
            else:
                print 'Failed to capture second finger'
        else:
            print 'Failed to capture first finger'
    else:
        print 'Failed: enroll storage is full'

if __name__ == '__main__':
    #changed baud rate to 115200. 09032015
    fps = FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=115200,timeout=2.0,is_com=False) #settings for raspberry pi GPIO
    fps.UseSerialDebug=False
    FPS.delay(2.5)
    if fps.SetLED(True):
        LegacyEnroll(fps)

cheatcodex, 2015-09-28:
indentation is completely messed up.
also, 
 if __name__ =='__main__':

ribs85, 2015-09-28:
Hi,

I gave up on this library, and am now using this one instead which uses node.js:
https://www.npmjs.com/package/gt511c3
Follow these instructions to download a version of node.js for the pi that actually works, otherwise you're going to have some serious headaches:
https://gist.github.com/justinjenkins/16adc15338bb265f6808

You can then use the supplied example to test everything, enroll your fingerprints and then test identifying. Make sure you edit the example to set the baud to 9600 and to change the /dev/ entry to what the pi uses. When enrolling with the example code, be sure to start enrolled, then p for fingerprint read, then enroll 1, then p, then enroll2, then p, then enroll 3. Make sure to remove and replace the *same finger* before each instance of pressing p. You can test the enrolment using "u".

I've never touched node.js before, but after a day of learning and messing around with the example script provided, and merging it into my own project, I now have something that works very well indeed. I used the node.js code to enroll and test all of my fingers. So far not a single false positive, and just a few (maybe 2%ish?) false negatives. False negatives are fine for my use case, as long as they are minimal. False positives are simply no good at all, thus far not got any of those.

I really wanted to like this library, but sadly it simply doesn't work for us pi users yet. Even after enrolling using the alternative library, this one still returned a result of 0 or 200 when trying to identify. I'm making the absurd assumption that this would work fine on a more vanilla desktop computer, as I assume that's what the author used. I have no idea why it would make a difference, but here we are.

I'm guessing someone smarter than me will be along to fork and fix this. But if you want something that "just works" today on a raspberry pi, use the node.js library instead for now.

cheatcodex, 2015-09-29:
I see. I'm not sure how to do multi-threading using node.js. I will defiantly give it a try. Just want to make sure, you are using Pi also?
I just wish to speak with the author quickly. I think fixes should be very minor and this can be up and running on pi. I felt that I've gotten really close to a working device. 
I will defiantly try out the one you recommended. Thanks!

cheatcodex, 2015-09-29:
Hi ribs85,
thanks for the advise. installing npm and the library took longer than the process of trying the code. I still have faith in this python library. I'll come back to it one day.

scriptmonster, 2015-10-27:
Hi, 

I know this issue is not active right now, but i think i figured out the problem. 
When Identify1_N method is called, it takes time to FPS to return response. When the empty response is returned from FPS module somehow returns 0. When i add 1 seconds of delay between SendCommand and GetResponse.

```python
    def Identify1_N(self):
        '''
             Checks the currently pressed finger against all enrolled fingerprints
             Returns:
                0-199: Verified against the specified ID (found, and here is the ID number)
                200: Failed to find the fingerprint in the database
        '''
        cp = Command_Packet('Identify1_N',UseSerialDebug=self.UseSerialDebug)
        packetbytes = cp.GetPacketBytes()
        self.SendCommand(packetbytes, 12)
        delay(1)
        rp = self.GetResponse()
        retval = rp.IntFromParameter()
        if retval > 200:
            retval = 200
        del rp
        del packetbytes
        del cp
        return retval
```

I tried lower values, to be more accurate i let it be a sec.  
There is a default delay in GetResponse method but it seems it's not enough.

I hope it would help.

cheatcodex, 2015-10-27:
Hi Scriptmonster,
I tried the same thing. Adding delays between the send and get functions. However, I discovered that different command needs different delays. Sometimes it takes 2 seconds or longer to get a response from the FPS, from what I've seen on the oscilloscope. 
I even went to 3 seconds at one point. I wasn't getting promising results either. However, the javascript library was a lot faster, which has me wondering why. 
Thanks a lot for your input! I have the device working with javascript now.

scriptmonster, 2015-10-27:
Hi cheatcodex,

I was having exactly same problem with you. I don't have an osciloscope but I have debugged the code and recognised the same thing as you. Even if you want to use GetRawImage method it would more than 20 seconds. But it works fine, makes your code wait untild the response is ready. The problem occurs when FPS response does not start immediately. After changing the code a little, i am getting reliable responses with the python library. I can read and identify a finger around 2-3 seconds.

If you know how much it takes to get response form Identify1_N method from node.js alternative, I would really like to know to compare performance.

I am developing an application which requires, GUI, GPIO and FPS. I have no idea if i could use node.js to access GPIO, and develop a local GUI thus i am sticking with python. I believe if the serial receiving approach is fixed, library could be more reliable and fast.

cheatcodex, 2015-10-27:
Hi Scriptmonster,
There are a lot of javascript packages. I just quickly searched for it. https://www.npmjs.com/package/pi-gpio
Not sure how good it is but I've been able to control a servo motor with js on the GPIO pins from a Pi. I was seeing somewere between 2.5s to 3.57s delays with this python library and the js library feels a lot shorter. identification feels like 500ms at most. I will measure it when I have a chance. I'm also exploring how to deal with the GUI part also. I'm trying to work out a way to display stuff on the LCD attached to the Pi.

eckama11, 2016-07-06:
I know many people probably think this is dead, but since I came across Josh Hawley's answer to this in CPP (I assume you took much of this from that collection since the comments are basically the same) I have made a simple observation about the logic. He implemented an example for enroll and part of it is CheckEnrolled. CheckEnrolled should return false if it is not enrolled and true if enrolled already. Yours assume the opposite even though CheckEnrolled looks exactly the same in the main program...