QuickCorp/pyGT511C3 #21: Why return packet data from get raw image is not 19200

thyahan, 2016-05-16:
Hi,
I've try to get fingerprint image and  I always got packet from get raw image command length 51854.
It could be 19200 because raw image size is 160*120, isn't it? 
do you know why?

closed 2016-05-16 by thyahan

deep4U, 2016-05-17:
Hi
@thyahan   I have similar confusionse regarding getrawimage please let us know how it was solved

thyahan, 2016-05-17:
Hi,
@deep4U Actually I didn't solved it, My return packet is not 19200 because my fingerprint sensor is GT511C1R and raw image is size is 240*216 [Datasheet](https://cdn.sparkfun.com/datasheets/Sensors/Biometric/GT-511C1R_datasheet_V1%205_20140312.pdf).
Hope this helps.

deep4U, 2016-05-17:
same happened with me I also have GT511C1R 
also do u know how to convert it into a image that we can view ?
and are you interfacing it with raspberry pi 2 successfully ?
Thanks

thyahan, 2016-05-17:
Yes, I am running on Raspberry pi 3 via UART, 
This is [example](https://github.com/QuickGroup/pyGT511C3/blob/master/test_raw.py)for convert to Bitmap.

deep4U, 2016-05-17:
I am facing lots of issues in making it work with my rpi2 can you help me through pm thanks

thyahan, 2016-05-17:
What kind of issues you're facing now, Could you write it here for someone has same issues like you.

deep4U, 2016-05-17:
I don't remember for sure as I tried and gave up long ago
Will redo the setup and let you know ASAP 
As well as I remember I got the connection but was getting random data printed and nothing basically worked 
I will setup , check and let you know
also did u directly connect it to rpi or some level converter / usb-uart was used?

thyahan, 2016-05-17:
I directly connect gt-511c1r to rpi3 uart without level converter and my path is /dev/ttyAMA0 
[I've read from this.](http://raspberrypi.stackexchange.com/questions/44176/connection-guide-for-gt-511c3-fingerprint-scanner-and-raspberry-pi-2-model-b)
```
1 Transmit connected to pin 10 (GPIO 15, RXD)
2 Receive connected to pin 8 (GPIO 14, TXD)
3 GND connected to a Pi ground pin (pin 6 )
4 Vcc connected to a Pi 5V pin (pin 4)

```

deep4U, 2016-07-05:
when I run example code I get all this
neither the cmos led turns on nor did I press my finger which was falsely detected
the same output comes always with same observation but my sensor works properly with arduino
GT511C1R is the one I am using and my board is raspberry pi 2 any Ideas what is wrong?

python ./raspberry_pi_GPIO_test.py 
Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
readed: 
Command: CmosLed
55 aa 01 00 01 00 00 00 12 00 13 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x12\x00\x13\x01
readed: 
Put your finger in the scan
Command: IsPressFinger
55 aa 01 00 01 00 00 00 26 00 27 01
U�&'
U\xaa\x01\x00\x01\x00\x00\x00&\x00'\x01
readed: 
Your finger is in the scan
Command: CmosLed
55 aa 01 00 00 00 00 00 12 00 12 01
U�
U\xaa\x01\x00\x00\x00\x00\x00\x12\x00\x12\x01
readed: 
Command: Close
55 aa 01 00 00 00 00 00 02 00 02 01
U�
U\xaa\x01\x00\x00\x00\x00\x00\x02\x00\x02\x01
readed:

thyahan, 2016-07-06:
Hello @deep4U,

LED must ON before captute finger or is press finger function.

**WHY LED DID NOT ON?**
I don't know exactly why, may be your GT511C1R baud rate was change (default 9600) so GT511C1R didn't receive command from RPI.
(GT511C1R first connect with 9600)
 
So you have to turn on LED before use IsPressFinger().
I recommend you to run this python's script below to check first.

```
import FPS, sys

fps =  FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=9600,timeout=2,is_com=False)

if fps.SetLED(True) :
    print 'LED OK'
    print 'Press your finger..'

    # Loop wait finger
    while not Fps.IsPressFinger():
        FPS.delay(1)

    print 'Fingerprint pressed!'    	

else:
    print 'LED FAIL'
```  

If you get LED FAIL or get error about connection fail, replug GND cable and GT511C1R will reset baud rate to default (9600).

deep4U, 2016-07-06:
yes I am getting LED FAIL
Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
readed: 
Command: CmosLed
55 aa 01 00 01 00 00 00 12 00 13 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x12\x00\x13\x01
readed: 
LED FAIL

but even after resetting baud I get the same 
while I am able to get it to work with arduino on same breadboard so I am sure no issue with connector or sensor 
any leads?

thyahan, 2016-07-06:
In Arduino is work, that mean GT511C1R didn't receive command from RPI.

I guess /dev/ttyAMA0 is not GT511C1R, I can't remember what device's mapping with /dev/ttyAMA0 by default but you have to free UART before use /dev/ttyAMA0

If your did't free uart on rpi before, Following this.
([Thank this guy](http://raspberrypihobbyist.blogspot.com/2012/08/raspberry-pi-serial-port.html))

Backup and edit /boot/cmdline.txt
```
sudo cp /boot/cmdline.txt /boot/cmdline_backup.txt
sudo nano /boot/cmdline.txt
```
Remove the two options referring to the serial port.
So, this

> dwc_otg.lpm_enable=0 **console=ttyAMA0,115200 kgdboc=ttyAMA0,115200** console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait

Becomes this

> dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait

You have to disable the serial getty service
([Thank this guy](https://www.raspberrypi.org/forums/memberlist.php?mode=viewprofile&u=162330))

With
```
sudo systemctl stop serial-getty@ttyAMA0.service
```
Disable from loading at boot
```
sudo systemctl disable serial-getty@ttyAMA0.service
```

Replug GND cable and again run this script.
```
import FPS, sys

fps =  FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=9600,timeout=2,is_com=False)

if fps.SetLED(True) :
    print 'LED OK'
    print 'Press your finger..'

    # Loop wait finger
    while not fps.IsPressFinger():
        FPS.delay(1)

    print 'Fingerprint pressed!'        

else:
    print 'LED FAIL'
```
If this all didn't work may be you have to try to reboot and try to reload library again.

deep4U, 2016-07-07:
hi
I tried all the things
I edited the file as suggested 
I executed the command also
but same result
than I tried replugging gnd still same
than rebooted 
still same
executed the commands to stop serial console still same output 
Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
readed: 
Command: CmosLed
55 aa 01 00 01 00 00 00 12 00 13 01
U�
U\xaa\x01\x00\x01\x00\x00\x00\x12\x00\x13\x01
readed: 
LED FAIL

I reverified that everything works fine with arduino and the sensor
There I am using a potential devider between rx of sensor and tx of arduino as arduino serial is of 5v
but for rpi I read the specs it is 3.3v uart but I am unable to get it to work
Any further inputs I can provide?
any log I shall send u ?

thyahan, 2016-07-07:
Hi @deep4U,
No, sent me nothing.

My GT511C1R didn't use any potential divider to connect with RPI3, just directly connected through UART.
I also connected with USB to TTL UART module, it properly worked.

Have you tried connect GT511C1R  to RPI without any potential divider?
Or Try with USB to TTL UART module?