QuickCorp/pyGT511C3 #22: facing LegacyEnroll problem

Tonyls93, 2016-05-22:
Hello Everyone. I'd like to ask for some help  (sorry for my english, I speak spanish)

I'm working with a Raspberry b+ and i'm using the UART GPIO with the baudrate = 115200. (I ran the script to modify the baudrate) 

At first I ran "raspberry_pi_GPIO_test.py" script and it worked fine ( CMOS led turns off when i put my finger on the sensor). Then I ran LegacyEnroll,py but the problem begins when it perform the "check enrolled" command, then it starts to check but the displays always finish saying "**Failed: enroll storage is full**". I also made a script to use the function "Delete All" but even by doing this, it didn't work.
[image]

This is the script I made to erase all ID's because i'm not sure if I wrote it in the right way

[image]

Thak you. I will appreciate your help

Mehluko, 2016-11-20:
HAE 
I hv the same problem did you find a solution for this

keyurrakholiya, 2017-01-26:
i am also facing same problem. please suggest if anybody find solution. @Mehluko @Tonyls93 @jeanmachuca

VenethKumar0808, 2018-01-10:
I get below error how to fix it

`
if baud <> self._serial.getBaudrate():
AttributeError: 'Serial' object has no attribute 'getBaudrate'
`