QuickCorp/pyGT511C3 #23: need help unable to use with rpi2

deep4U, 2016-07-05:
when I run example code I get all this
neither the cmos led turns on nor did I press my finger which was falsely detected
the same output comes always with same observation but my sensor works properly with arduino
GT511C1R is the one I am using and my board is raspberry pi 2 any Ideas what is wrong?

python ./raspberry_pi_GPIO_test.py
Command: Open
55 aa 01 00 01 00 00 00 01 00 02 01
U������
U\xaa\x01\x00\x01\x00\x00\x00\x01\x00\x02\x01
readed:
Command: CmosLed
55 aa 01 00 01 00 00 00 12 00 13 01
U������
U\xaa\x01\x00\x01\x00\x00\x00\x12\x00\x13\x01
readed:
Put your finger in the scan
Command: IsPressFinger
55 aa 01 00 01 00 00 00 26 00 27 01
U���&'�
U\xaa\x01\x00\x01\x00\x00\x00&\x00'\x01
readed:
Your finger is in the scan
Command: CmosLed
55 aa 01 00 00 00 00 00 12 00 12 01
U�����
U\xaa\x01\x00\x00\x00\x00\x00\x12\x00\x12\x01
readed:
Command: Close
55 aa 01 00 00 00 00 00 02 00 02 01
U�����
U\xaa\x01\x00\x00\x00\x00\x00\x02\x00\x02\x01
readed:

himanshu-hearthacker, 2016-07-09:
facing same problem, did you get any solution yet?

felipefuller, 2016-07-13:
I have the same problem! any solution?

felipefuller, 2016-07-14:
Found a temporary solution! after searching the whole internet, i installed de 2015-05-05 wheezy image from: https://downloads.raspberrypi.org/raspbian/images/ Then i used this code to try the sensor http://makesensors.botbook.com/code.html and worked :) Now im trying to use this library. I've had a some errors, but should work. I hope this can help you!

deep4U, 2016-07-18:
Hey guys sorry have been busy lately 
still facing same issues tried lots of things
gonna try @felipefuller 's solution this weekend and get back ASAP

deep4U, 2016-07-18:
I read the code from http://makesensors.botbook.com/code.html
for those having dificulties in finding it is in fingerprint_scanner as a python script
also I would remove these lines from the script's main before running 
print("Remove all identities from scanner")
and	removeAll()
as I want to just check the led is getting on or not. :-)
Will install the said version of wheezy on my pi and run this script to let you know 
thanks

felipefuller, 2016-07-18:
after trying that solution, i noticed the response of the FPS was very slow. It worked, but was very slow. I also removed the "removeAll()" and enrolled without problems. After trying any possible solution to get this library working, i gave a shot to the node.js library and worked surprisingly awesome!

Mehluko, 2016-11-20:
@ felipefuller how did you get the code from node.js script working I'm trying to use it I can get it working.w@ steps did you follow?

himanshu-hearthacker, 2016-11-21:
replace FPS.py file with the file named FPS2.py [https://github.com/Jucker/SIPIntercom-with-FPS](url). this is guy modified the file to work with code. if you face any problem try to change the baud rate 2-3 time.