QuickCorp/pyGT511C3 #25: using wiith Rasberry PI 3 no response

ColeBracew, 2016-10-12:
Hi i have hooked up the scanner correctly to my GPIO board and when i try run  	raspberry_pi_GPIO_test.py all i get is Command:open then

Traceback (most recent call last): File "/home/pi/Desktop/pyGT511C3-master/raspberry_pi_GPIO_test.py", line 32, in <module>
    fps =  FPS.FPS_GT511C3(device_name='/dev/ttyAMA0',baud=9600,timeout=5,is_com=False)
  File "/home/pi/Desktop/pyGT511C3-master/FPS.py", line 319, in __init__
    self.Open()
  File "/home/pi/Desktop/pyGT511C3-master/FPS.py", line 333, in Open
    self.SendCommand(packetbytes, 12)
  File "/home/pi/Desktop/pyGT511C3-master/FPS.py", line 756, in SendCommand
    self._serial.write(bytes(cmd))
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 475, in write
    n = os.write(self.fd, d)
KeyboardInterrupt when i inturupt to quit out 

 if i try to run baud_to_115200.py i get the following: 
 
Traceback (most recent call last):
  File "/home/pi/Desktop/pyGT511C3-master/baud_to_115200.py", line 16, in <module>
    fps = FPS.FPS_GT511C3()
  File "/home/pi/Desktop/pyGT511C3-master/FPS.py", line 316, in __init__
    self._serial = connect(device_name,baud,timeout,is_com=is_com)
  File "/home/pi/Desktop/pyGT511C3-master/FPS.py", line 277, in connect
    device_name = DEVICE_NAME
UnboundLocalError: local variable 'DEVICE_NAME' referenced before assignment

Can anyone advise? I'm just getting into PI's and i don't have a firm understanding. Thanks

sandeep123patel, 2016-10-16:
I have the same problem! any solution?

himanshu-hearthacker, 2016-11-21:
replace FPS.py in your code with file named FPS2.py. https://github.com/Jucker/SIPIntercom-with-FPS .This is guy modified the file to work with code. if you face any problem try to change the baud rate 2-3 time

jeanmachuca, 2016-11-21:
Hi folks, 
This driver has a paid support via Skype. You just need to donate USD 100 on the link bellow once, and you will get all the online support needed to make your implementation works fine:

https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=Q4L8KWJTYL5NY

Cheers,

Abhiramianbu, 2017-05-31:
Anyone found the solution ? I have the same problem

Abhiramianbu, 2017-06-08:
For me the problem is solved. The serial port should be enabled for connecting devices .By default it is used for bluetooth.

Druffl3, 2018-04-21:
@himanshu-hearthacker Man, even after changing baud rates, I'm getting 
"incomplete response in buffer".
Please, Help.