QuickCorp/pyGT511C3 #30: code for capturing finger print image

Abhiramianbu, 2017-09-09:
somebody have any program for capturing and saving the image in jpeg format?

athachirag, 2017-09-09:
Use This :  https://github.com/zafartahirov/fingerpi
I have worked on GT511C3-master Finger print scanner.
This code is completely working.
Note: Do take care of the GPIO pin configuration of the scanner and match the pin configuration in the code

athachirag, 2017-09-09:
Can you please elaborate about the scanner and controller/processor you are using???

Abhiramianbu, 2017-09-09:
Procesor - Raspberry pi3
Scanner - ttl gt511c3

athachirag, 2017-09-09:
okk great,
Use This: https://github.com/zafartahirov/fingerpi
I have used this before on Pi 3
It is completely working for GT511C3 
Note : Do take care of the gpio configuration of the scanner and the gpio configuration used in the code

MarletTorrontegui, 2017-12-15:
Hello! 

I'm having some troubles, can anyone provide me some information about the gpio configuration, I'm new using Raspberry. 

Hope anyone can help me. 

Regards!