QuickCorp/pyGT511C3 #31: 'Serial' object has no attribute 'getBaudrate'

Nipun-Bhalla, 2018-04-22:
Here is the error I keep getting:

`if baud <> self._serial.getBaudrate(): AttributeError: 'Serial' object has no attribute 'getBaudrate'
`
Using GT511C3 with a USB to TTY converter on RPi3.

Pretty sure it has something to do with pySerial. Currently using pyserial 2.7

wamburu, 2018-08-07:
Am also running into this error, any work around it?

miqi95, 2019-01-22:
I also have this problem, using GT521F52 on the Tx/Rx pins of an RPi3. Commented on the most recent commit.

miqi95, 2019-01-22:
I changed the line in FPS.py from self.serial.getBaudrate() to self.serial.baudrate. That error is gone, but now it just says "Command: Open" and freezes up.

meie1kyl, 2019-02-14:
@miqi95 Have you made any progress on that issue? I'm also at the same point where it only says "Command: Open"

miqi95, 2019-02-15:
@meie1kyl No, given the lack of support I decided to switch to the U.are.U 4500 scanner which is supported by [fprint](https://www.freedesktop.org/wiki/Software/fprint/) for my application. Coupling that with the python 3 subprocess module I'm able to use any scanner supported by the driver, plug & play.